machineagency/jubilee #3: Jubilee's Bed plate (with BuildTak) takes ~5-to-8 minutes to finish thermally expanding

Poofjunior, 2019-10-30:
Although the silicone heater can reach 60C from ambient in about 1 minute, the combined thermal mass of the aluminum bed and the Buildtak magnetic plate take about 5 minutes to finish thermally expanding. During this time, the printer will expand vertically about 0.1mm--enough to make-or-break a first layer.

If the machine is zeroed while the bed is cool and the print is started before the machine has finished thermally expanding, the first layer risks ripping as the bed-to-nozzle distance will eventually become too close to the bed.

A first pass at fixing this issue would be to zero the Z height at the start of the print after bed has reached temperature. However, since the sensor is on the _underside_ of the machine, this does not provide an accurate estimate of the temperature of the entire plate at steady state.

A slightly-better fix would be to decrease the thermal mass of the aluminum bed plate. This fix is being tested now as a future revision of the Jubilee bed plate will feature magnets embedded in the build plate. This updated plate will remove the added thermal mass of the Buildtak magnetic surface.

Another slightly-better solution would be to measure the temperature closer to the print surface. The Railcore bed plate puts the bed sensor in a pocket on the top of the build plate, not the bottom. Jubilee could adopt this solution.

For now, the surefire way to work with this issue is to (1) heat the bed, (2) wait 5 minutes, (3) zero the z and begin printing.

This issue will get an update as soon as the new beds with smaller thermal mass are tested.

KodinLanewave, 2020-01-20:
Why not modify the design by drilling a hole in the center of the bed plate, (prior to putting any surfaces on it, and maybe countersink both sides of the hole to eliminate burrs, etc...) then submerge the thermistor in thermally conductive epoxy?  Might give you a better indication of the plate temperature rather than just the silicone heater temperature.

Poofjunior, 2020-01-20:
Aye; this is similar to the last suggestion. It's a bit of extra work that I'd love to avoid burdening the user with if possible. I'll have some more updates on this issue this week, but it also seems like the embedded magnet plate heats up wayy faster (less than 4 minutes).

KodinLanewave, 2020-01-20:
No one has done a spring steel (coated or otherwise) build surface with integrated thermal sensor(s) yet, right?

Poofjunior, 2020-01-20:
Hmm, Not that I know of. Are you envisioning a particular sensor in mind?

KodinLanewave, 2020-01-20:
No, just spitballing ideas; For safety you'd want to still monitor the sensor in the silicone heater, (especially if you're using one of the higher wattage A/C heating pads due to the fire danger) but the critical temperature measurement for printing is as close to the print surface as possible; why not have one directly integrated into the print surface if you're going so far as to design an entire machine?  Just a thought, no idea how practical it'd be to implement.  Still think the 1/4" hole in the middle of the thicker build plates make more sense both from a implementation standpoint and a design standpoint.  CAD files can be updated for those that are making bed plates, anyone who already has one probably either owns or can gain access to a drill and drill bit and some JB-Weld epoxy.

Poofjunior, 2020-01-20:
Gotcha. Yeah; before I go the "invasive" route, I'm going to start with a bolt-on terminal lug thermistor.
https://www.digikey.com/product-detail/en/vishay-bc-components/NTCALUG01A103F/BC2891-ND/4896928
This seems to be an industrial standard for measuring plate temps that I haven't seen (yet) on printers. Best of all, it can be screwed into an existing threaded hole.

I plan on checking how well both thermistors track against some "ground truth" measurements from the top.

KodinLanewave, 2020-01-20:
Oh interesting. Might want to add some silver thermal paste, (or equivalent,) between the mating parts to minimize conduction losses.

Poofjunior, 2020-01-20:
Stay tuned, and I'll keep everyone posted on how it goes!

Poofjunior, 2020-01-23:
Ok I'm calling this a mystery solved.

Using an IR gun to monitor the top plate temperature, I ran two setups. Both had
* magnetic plate
* keenovo heated pad
* calibrated with the thermistor they were using

one had the *built-in* thermocouple and one had the *bolt-on* thermocouple.

## Built-In Case
The built-in thermocouple measures 21C to 60C in 1:15, but at the 1:15 mark, the IR gun measures the top surface to still only be 45-47C. After 5 minutes, the built-in thermocouple and IR gun agree at about 60-61C

## Bolt-On Case
The bolt-on thermocouple measures 21C to 60C in about 2:15. The plate temp overshoots by about 3-5C at this time but cools to 60-61C in about 5 minutes.

## The Verdict:
Use the bolt-on thermocouple. Even though it lags behind the true plate temperature, it guarantees that the plate wont keep expanding when a print has started. Waiting 120 to 180 seconds is also super quick by comparison with most printers.

Bonus: it's like $2-3 and detachable, so it's easy to replace.

The BOM and instructions already point to using the embedded mag plate and the bolt on thermocouple, so I'm marking this issue resolved.

closed 2020-01-23 by Poofjunior