machineagency/jubilee #4: Default tool Limits maximum bed size to 290 in Y

Poofjunior, 2019-11-01:
This picture says it all. 

https://media.discordapp.net/attachments/627988602269335568/639674827103273000/IMG_0850.jpeg?width=893&height=670

Essentially, the default tool's thickness limits the prin area in the y direction to about 290mm in a multitool setup.

DanalEstes, 2019-11-01:
I believe this can be resolved by making the right and left rail of the upper frame longer, thus moving out the "upper front" rail that carries the tools, moving it in the Y direction.  It looks like we need about 15 to 20 more mm. Relocate the front R/L pulley assemblies by the same amount. That implies longer belts. I don't believe any other part of the printer would have to change, in a "retrofit" situation. Not even the linear slides.

solhuebner, 2020-02-09:
@DanalEstes As this was implemented in 2.0, can this issue be closed?

Poofjunior, 2020-02-09:
@solhuebner aye!
This is fixed as of rel/2.0
https://github.com/machineagency/jubilee/blob/master/CHANGELOG.md#reljubilee-20---jan-15-2020

closed 2020-02-09 by Poofjunior