machineagency/jubilee #8: Where is the Top Lock Actuator Plate?

vcapra1, 2019-12-26:
In the assembly instructions (page 2 of [toolchanger_lock_assembly_instructions.pdf](https://github.com/machineagency/jubilee/blob/master/assembly_instructions/toolchanger_mechanism/toolchanger_lock_assembly_instructions.pdf)), it says a "Top Lock Actuator Plate" is needed.  There is no such part in the list of 3d printed parts.  The closest thing I can find is the "Lock Actuator Base Plate", which is referenced on the following page and does in fact exist in the list of 3d printed parts.

The image of the Top Lock Actuator Plate does not match the 3d model, but the model does match the Lock Actuator Base Plate on the following page.  In the first case, there are 2 more holes and 2 slots in the side of the part, which don't exist on the next page, or the 3d model.  Am I to assume these are in fact the same part, and those holes/slots are not necessary?

timcurtis67, 2019-12-27:
I believe the part you are looking for is really called "top_lock_actuator_plate_Spring Guide Anchor Block.stl" or .sldprt. Here is a copy of that file.

All of the holes are used, the extra holes you see are to bolt this part to the base plate.
[top_lock_actuator_plate_Spring Guide Anchor Block.zip](https://github.com/machineagency/jubilee/files/4005157/top_lock_actuator_plate_Spring.Guide.Anchor.Block.zip)

vcapra1, 2019-12-28:
That is not what I meant.  The assembly instructions show the following part:
[image]
But this is the actual 3d model of that part:
[image]
As you can see, there are two holes missing from the 3d model, as well as the ziptie slots. on the side.  These holes are where the spring guide anchor block is mounted, so now I can't attach that piece because the holes are missing.

vcapra1, 2019-12-29:
I'm closing this issue as it appears the guide block isn't necessary, so the missing holes are not a problem.

closed 2019-12-29 by vcapra1