machineagency/jubilee #9: Extruder Nozzle does not extend below tool plate

vcapra1, 2020-01-02:
I think this picture pretty much sums it up:

[image]

I know it would be pretty easy for me to design an adaptor for this, but I figured I'd put something here since these are the parts listed on the BOM

Poofjunior, 2020-01-07:
Hmm, there are definitely missing parts in this assembly. Check the latest version of the files on the development branch:
https://github.com/machineagency/jubilee/tree/dev/jubilee_2.0/cads/tools/direct_drive_bondtech_groovemount_extruder

Poofjunior, 2020-03-17:
The extruder now has assembly instructions. Hopefully, that clears up the order of assembly.
https://github.com/machineagency/jubilee/blob/master/tools/jubilee_tools/tools/extruders/direct_drive_bondtech_groovemount_extruder/assembly_instructions/bondtech_groovemount_extruder_assembly_instructions.pdf

closed 2020-03-17 by Poofjunior