machineagency/jubilee #11: BOM - Part numbers w/ ver. Embed part number in printed parts

BryanHaven, 2020-01-08:
Include a field in the BOM for part numbers
Part identifier should reference the assembly the part is used in
Part numbers should have version reference
Engrave/emboss part number in printed parts so when printed part number is visible
Add part numbers to machined parts - fabricator may determine best method for marking of part (laser etch, stamped, machined, etc.)

Poofjunior, 2020-01-23:
Ok, we currently have a Shopping list, not a proper BOM, but I can export one pretty easily, that we can group into subassemblies.

Itemizing from above, it looks like part numbers should have:
* some cross-reference to the sub-assembly that it belongs to.
* unique numerical identifier
* part revision
* part process (*P*rinted, *L*aser cut, *M*achined, *V*itamin etc.)

Here are some special cases to consider:
* interchangeable parts. Is there a way we can quickly identify by part number whether or not something is a replacement for something else? Example: a metal crossbar and an aluminum crossbar are functionally the same thing.
* Parts that are configurable. Example: on the parking posts, the width of the unit is parametric, enabling us to generate a range of sizes depending on the tool. We might be able to tack on the parameter value at the end of the part number.

BryanHaven, 2020-01-23:
https://grabcad.com/tutorials/3d-printing-design-guidelines
Posting reference for lettering best practices - mostly so I don’t forget.

laudiblellc, 2020-01-23:
I would recommend including the part dimensions' dependence on printer
size, or at least print volume size. The hypercube Evo bom does this and
makes customizing your printer very easy.
Saul

Poofjunior, 2020-01-23:
@BryanHaven and @laudiblellc I think you're both getting at "part  printability" with your later discussion points. Some of these are actually already called out in the [Design Objectives](https://github.com/machineagency/jubilee/wiki/Project-Goals#design-objectives) doc, but I could flesh out some details such that others can contribute back parts that are modeled in the same spirit.

BryanHaven, 2020-01-23:
@Poofjunior True, primarily I woke up vaguely remembering something I’d read and wanted to get it down somewhere useful. In this case specifically related to etching or embossing text into a printed part. It’s easy to make the part numbers legible on screen but harder when you print them.

Poofjunior, 2020-01-29:
Ok, here's a pass at a part numbering scheme for all non-stock parts.

**[who designed it] - [fab process + material] - [uniqueid] - [revision] - [who fabricated it]**
* **[who designed it]** 3 characters. I'll claim **PJR**, short for poofjunior. This is a nice way to credit folks. If we add stuff to the repo, they get a part number that references them. It also helps us trace design-related bugs and points us to who we can ask for help debugging. If mulitple people design it, we can use **MIX**
* **[fab process + material]** 3 characters
    * **M**achined
        * **M6** MIC6 aluminum
        * **61** 6061 aluminum
        * **K1** K100S aluminum
        * **A5** ATP5 aluminum
        * **C**arbon **F**iber
    * 3D**P**rinted
        * **PL** PLA
        * **PE** PETG
        * **CP** **C**arbon filled **P**LA
        * **CE** **C**arbon filled P**E**TG
    * **L**aser cut
        * **DE**lrin
        * **AC**rylic
        * **PL**ywood
* **[unique id]** 3-9 characters. 3 preferrable. Up to 9 to denote special configurations/mirrors
* **[revision]** 2 characters revision number
* **[who fabricated it]**
    * **713**Maker
   * **C**NC **M**a**D**ness
    * **S**ee**M**e**C**NC
    * **M**andala **R**ose **W**orks
    * **MIS**umi
    * **YOU** make it yourself by default. (3d printed parts)

Here are some examples:
* PJR-MCF-CRO-01-CMD
    * [poofjunior] designed a [Machined] [Carbon Fiber] [Crossbar] [revision 1] manufactured by [Mandala Rose Works]
* PJR-MK1-CRO-01-MRW
    * [Poofjunior] designed a [Machined] [K100S] [Crossbar] [Revision 1] manufactured by [Mandala Rose Works].
* MIX-MM6-BED-01-713
    * [Poofjunior and 713Maker] designed a [Machined] [MIC6 aluminum] [bed] [revision 1] manufactured by [713Maker]

This numbering scheme has some tradeoffs. The main one is that it doesn't have a reference to the subassembly. I don't think we *really* need it since we have less than 10 top-level subassemblies and we can almost-certainly deduce what fabricated part was made based on the unique id.

But it also has some major benefits, I imagine that folks will build a Jubilee with parts sourced from various places. Knowing how someone configured their particular Jubilee is extremely useful for helping us figure out how to help them debug an issue. For example: I imagine that the aluminum crossbar will have a different top speed than the carbon crossbar. I also imagine that a carriage made from PLA will have different maximum temperature rating compared to PETG. What's more, we can also invert the supply chain and ask them to stock specific part numbers. (Example: we could ask CNC Madness to stock PJR-MCF-CRO-01-CMD. That way people who want a carbon crossbar can just ask for PJR-MCF-CRO-01-CMD instead of emailing him a DXF file with material specs.)

Ok, what do people think?

Enza3d, 2020-01-29:
I like the content of the above part #'s 
I would argue the Creator of the part should go on the end of the part number to aid in listing items alphabetically in a coherent fashion.  Ideally the most relevant piece of information is listed as the first thing in the string so it can be sorted by that parameter.

At work we start our Main assemblies with 00 so it always appears as the first item in a list. The sub assemblies are followed by 01,02,03 etc... 
Then the parts are 101,102,103... for the parts in the 01 sub assembly.  201,202,203 for parts in the 02 assembly. 

Since we do consulting work every project has an Identifier code which syncs our sales team with the engineering team.  This 4 Digit number is at the front of every single project file we create.   So a full main assembly would look like 4010-00.SLDASM, a part would look like 4010-105.SLDPRT

We actively chose not to capture any other information in the part number to keep it light weight and easy to understand.  We dont do any real fabrication or sourcing though so we dont need to bake as much info into our Part #'s.

tsunaminaut, 2020-01-29:
Agreed about using main assemble numbers first then part of assembly second, I use a similar scheme when saving media i label them as last 2 digits of year, followed by month and then date so when viewed in a file they will organize by year then month then date and be in a logical order and easy to find.

Enza3d, 2020-01-29:
I will add, the simplicity of a part numbering scheme is often directly corelated to the compliance, if its too complicated (need a decoder ring to unscramble) people will tend to "forget" to use it, where if its just the bare minimum people will follow it.  The whole thing should be easy to memorize and simple to figure out without a Key

BryanHaven, 2020-01-29:
I’d also add that if we’re looking at etching / embossing the part number into the printed and machined parts they shouldn’t be but so long. A vendor cross-reference chart may be a solution to the simple part number for the project and vendor specific SKUs.

Poofjunior, 2020-01-29:
Thanks for the feedback!

To trim length, I can probably drop the "who-designed-it" field and re-order the fields so that the initial bit is more relevant so it looks like:

**[uniqueid] - [revision] - [fab process + material] - [who fabricated it]**

In cases where machinists add their own insignia, we can drop the last bit so it looks like:

**[uniqueid] - [revision] - [fab process + material]**

Finally for printed parts, we could display just embed the first two bits:

**[uniqueid] - [revision]**

As far as unique id goes, **[subassembly] - [number]** is one option. Some parts have configurations, though, and it would be nice to capture some of the configuration parameters in this section. Here's an example.

The tool holder is parametric, where the spacing is configurable to handle wider or skinnier tools. The default has 55mm center-to-center hole spacing.
[image]

But we could easily change that to be 65mm:
[image]

Anyone have any thoughts on configurable parts?

I would like the part numbering scheme to capture this configurability in the unique id somehow, and I think something more flexible than **[subassembly] - [number]** would be useful here.  Maybe **[subassembly] - [number] - [config params]**, but I don't have a good answer for this just yet.

Finally, one last thing to note: in this case, I want the part number to serve the community more than serve just the designer. For that reason, while having a things like a project number to prefix the part number would be useful to me, it's extraneous information for the person trying to build the machine or modify existing parts. I think keeping the part number short is one solution to this. But the big problem is creating a convention that serves people who will be assembling a machine that is being tweaked with revisions over time so that they have a clear idea of what their configuration or version of the machine can do, and how it compares against the latest version.

Enza3d, 2020-01-29:
We add a two digit code at the end of our part numbers which denotes the configuration, so 3333-101-00 denotes the default config, 3333-101-01 denotes the first configuration of part 101.   

 I dont particularly care for this approach but we haven't been able to come up with something better.

Poofjunior, 2020-01-30:
Ok, I have an updated draft that incorporates sub-assembly numbers. How does this look?

## Part Numbering Scheme
The part numbering scheme is as follows:

**[subassembly] - [part id + params] - [revision] - [fab process + material] - [who fabricated it]**

For 3D printed parts, only the short numbering scheme will be used.

**[subassembly] - [part id + params] - [revision]**

Here's the breakdown of what these mean.

* **[subassembly]** what lowest-level subassembly does this belong to?
    * 00 top level assembly
    * 02 motor corner plates
    * 03 double pulley corner plates
    * 04 toolchanger elastic lock assembly
    * 05 electronics back panel
    * 06 side panels
    * 07 toolchanger carriage component assembly
    * 09 Build Plate Assembly
    * ?? Lower Frame Assembly
    * 01 corexy frame assembly
    * 08 Z axis assembly
* **[part id + params]** 3-9 characters. (Uppercase) letters and numbers only. Try to keep these around 3 characters while the rest are reserved for parameters. Here are some common ones.
    * **EXT409** -- extrusion of length 409mm
    * **CRS** -- crossbar
    * **BED** -- bed plate
* **[revision]** two digits increasing from 0
* **[fab process + material]**
    * **M**achined
        * 60**61** aluminum
        * **K**100**S** aluminum cast tool plate
        * **M**IC**6** aluminum cast tool plate
        * **A**TP**5** aluminum cast tool plate
    * 3D**P**rinted
        * **PL**A
        * **PE**TG
        * **C**arbon Fiber impregnated P**L**A
        * **C**arbon Fiber impregnated P**E**TG
    * **L**aser cut
* **[who fabricated it]** This can be an open field in the scheme that can be used by the machinist. Otherwise, here are some defaults.
    * **M**i**SU**mi
    * **713**Maker
    * **S**ee**M**e**C**NC
    * **M**andala **R**ose **W**orks

### Blacklisted Letters
Let's not use these letters: i, o

This is also in a draft here:
https://github.com/machineagency/jubilee/wiki/Part-Numbers

Kirinkunhime, 2020-02-01:
I want to suggest a minor addition to this: And that is that all part-numbers should be written with capitalletters only. (As is already implied)

This should help improve readability, and also the letter l might be too similar to a 1 otherwise.

Poofjunior, 2020-02-02:
Thanks @Kirinkunhime! Adding this to a note on the bottom of the current doc.

Poofjunior, 2020-02-10:
Ok, I think I managed to go through and capture most suggestions here. Short part numbers (with revision IDs) are now embedded in their respective parts for printed parts. Full part numbers are specified in the wiki doc linked above.

These are driven from metadata attached to each part, so exporting these to a spreadsheet in the future should be pretty straightforward.

Poofjunior, 2020-02-10:
Oh, here are some examples:
[front_left_foot](https://github.com/machineagency/jubilee/blob/master/frame/fabrication_exports/3d_printed_parts/frame/front_left_foot.STL)
[right motor plate](https://github.com/machineagency/jubilee/blob/master/frame/fabrication_exports/3d_printed_parts/frame/right_motor_plate.STL) (see underside)

closed 2020-02-10 by Poofjunior