machineagency/jubilee #19: Outer Frame Assembly and Alignment Instructions

cindygfeng, 2020-01-29:
Sheet 2: We had some confusion on which way the brackets go in.  Include a statement that says that the interior inside corner brackets are directional, and they go on the 604 mm TSlot Extrusion first

[image]

Poofjunior, 2020-01-29:
Sure; I can change the first word chunk to say:

_First slide the brackets into the long extrusions. Then fit the remaining parts together as shown above without tightening down any screws. Lay the loose assembly on a flat surface._

_Note that only one side of the bracket is meant to completely slide into the extrusion._

Poofjunior, 2020-01-30:
[image]

How about this?

Poofjunior, 2020-02-20:
Closing; I think this should be sufficient.

closed 2020-02-20 by Poofjunior