machineagency/jubilee #26: Warping Extruder Tool Plates

Poofjunior, 2020-01-30:
It's unclear whether this is coming from the hotend or the bed, but the excess radiated heat is causing these to warp over time. 

[image]

They seem to settle after a bit which is why I never noticed this and probably just recalibrated the heads months ago.

These are made from PLA by default. I will start by giving these a go in PETG.

Poofjunior, 2020-02-10:
Reprinted from PETG.

It turns out that the source of error appears to be a rather hot extruder stepper motor, which will affect even the PETG plates. However, adding a [heatsink](https://www.amazon.com/gp/product/B07FNK2866/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) to the back appears to drop the temperature down to about 48C and prevent further warping.

I will continue to run the machine in this state for a bit longer before calling it a problem-solved.

Poofjunior, 2020-02-20:
This is definitely due to a hot extruder motor. Printing material doesn't really matter as long as the heatsink is applied. The heatsink is now in the extruder BOM.

Poofjunior, 2020-09-09:
It looks like ambient radiation is still having an effect on tool plates. Here's a PLA plate built to the current spec by @guildner 

[image]

matthewSorensen, 2020-09-09:
Definitely still an issue, and seemingly independent of the heater block orientation. Since it's not a universal complaint, I wonder how much it depends on the specific filament the plate was printed in, and the workload of the tool?

[image]

guildner, 2020-09-09:
Additional information: my bed temperature was at 80C, and the extruder was at 235C. The print job was long, like 13 hours, and flat and close to the bed surface for a large portion of the print time. (large flat objects that took up most of the build plate.)

Poofjunior, 2020-09-09:
I think the remaining causes on this issue could be:

- ambient radiation from the hotend or bed
- creep from the axial force applied by the twist lock
- the combination of both

A metal plate will fix all of these issues. That said, it it's just ambient radiation, it would be nice to have some sort of heat shield (tape) solution that wouldn't necessitate metal parts in the minimum viable design. That said, maybe I'm asking too much since we're dealing with a system that's producing heat to melt plastic precisely in the first place ;)

guildner, 2020-09-09:
@matthewSorensen what materials (bed temp) do you usually use? Did yours warp after a particularly long or hot print? Or is that something that progressed over time?

This may not be as much of a problem for lower bed temperatures, or prints that are only close to the bed for a shorter amount of time... just thoughts.

Blade765, 2020-09-10:
What about just using a material rather than PLA? Maybe even heat treated HTPLA? I think the deflection temp after heat treating is 150ish? Granted, requiring a special material for the plates is an added expense, but it would still be significantly cheaper than metal.

Blade765, 2020-09-11:
I was referring to Proto-Pasta's Carbon Fiber HTPLA. Once heat treated, its good for up to ~160*C.

Poofjunior, 2020-12-03:
Ok; the current default points to getting a [metal version of this plate](https://shop.guildtek.com/index.php?main_page=product_info&cPath=1&products_id=4).

I think it's time to close this issue and reopen another one asking for a simpler fix that doesn't necessitate such a complicated metal part. I don't want to jump to far and mandate metal plates for tools, but this design should hold us over at least until we come up with something better to replace it.

closed 2020-12-03 by Poofjunior