machineagency/jubilee #31: Lower Frame Assembly instructions - Z Plate Tightening

kelliead, 2020-02-05:
Lower Frame Assembly instructions, sheet 13 of 14, Z Plate Tightening

Add dimensions for correct placement of back Z plate using a ruler. 15.5cm from edge of plate to inside edge of side extrusion.

[image]

Poofjunior, 2020-02-06:
Ok, so it turns out that I did add this info--I just never exported it--doh!

Poofjunior, 2020-02-06:
@kelliead @awaer @cindygfeng actually this is already fixed and pushed online. make sure you're working off the latest instructions on the web, not a local copy that you downloaded.

closed 2020-02-06 by Poofjunior

cindygfeng, 2020-02-08:
@Poofjunior We were working off of instructions that we printed that day from the Assembly Instructions wiki page.  Perhaps the link hadn't been updated, and that's why we had old instructions? @kelliead