machineagency/jubilee #43: Inconsistent Tool Pickup

Poofjunior, 2020-02-15:
It's possible (and actually quite likely) that the tool can be picked off the parking post in such a way that doesn't properly lock the tool into the carriage. This is because the twist lock can cause the entire to tool to twist if it snags in the wedge before pulling it into the coupling. The result is that the tool balls don't seat against the pins and are instead constrained by the wedge.

### Diagnosis:
This issue is nefarious as it can be hard to track down without an upward-pointing webcam
* Monochrome prints look perfect, but multicolor prints look like they have inconsistent extrusion issues.
* The tool may physically "jiggle" in the carriage if you try to wiggle it with your hands. However, subsequent pickups may or may not jiggle.

### Possible Fixes
There are a number of ways to fix this problem:
* stiffer spring (+ possibly a stronger motor)
    * this would overpower the twist applied to the tool and eventually get it to seat properly into the coupling.
    * this fix can be tried by shortening the current length of the spring a few mm. 103.5-to-104mm instead of 106.5mm
* better parking post
    * this would prevent the current tool from twisting as it is being picked up.
* better tool constraint upon tool pickup
    * when the tool is picked up, something could physically prevent the tool from twisting upon pickup.
        * This could simply be a parking post that prevents twist.
        * This could also be some external nesting force (spring, opposing magnets) that pushes on the tool as it is being picked up.
* smoother Wedge Plates
    * this would prevent the tool from snagging on the wedge plate and twisting as it is being picked up.
* (Other suggestions welcome!)

I'm opting to start with the second option. The current parking post is admittedly pretty crappy in that it's basically a giant pivot point

BryanHaven, 2020-02-15:
Have you updated the printing instructions for the lock plate insert to print at the lower layer height?

Poofjunior, 2020-02-15:
Oof, good point. I completely dropped the ball on extruder instructions. I just started a page (draft!) for them.

https://github.com/machineagency/jubilee/wiki/Bondtech-Direct-Drive-Extruder

DanalEstes, 2020-02-15:
I did notice the finer layers on V1 instructions and did print the plates that way.  I've noticed this same thing, and with regard to the 'wiggle', sometimes when you first touch the mounted tool it will 'pop' into place and won't wiggle after that. 

At this moment, I've switched back to stainless pins, and silver balls (i.e. original spec).  I am also going to shorten my spring a bit for more torque (mine may be the slightest bit long to begin with). 

I will report back when there are a few hundred or thousand tool exchanges with these updates.

Poofjunior, 2020-02-16:
@DanalEstes have a go at the [new parking post design](https://github.com/machineagency/jubilee/tree/master/tools/jubilee_tools/tool_posts/configurable_tool_post). Still need to run more than just one test, but the results look very promising. (Results at the tail end of this related issue: https://github.com/machineagency/jubilee/issues/43 )

Poofjunior, 2020-02-20:
Switching to the new parking post seems to have fixed this issue since tools can no longer easily twist while being picked up.

vcapra1, 2020-04-04:
I was having the same issue, and I'm actually using aluminum parking post blocks so there's zero room for flex there.

I tried shortening the actuator spring incrementally, but I had to take off around 6mm (total length around 100mm) before seeing any improvement.  At that point, it was extremely difficult to assemble the actuator pulleys so I didn't want to go any further with that, and the problem hadn't fully gone away.

I then tried slowing the motor down to about 50% of it's normal speed for locking (as I now see you've already updated the macro file with).  I also sanded the ramp on the wedge plates to make them smoother, but printing at a smaller layer height would also work for that.

Just figured I'd share what helped solve the issue for me in case anyone else is still experiencing this problem.

Poofjunior, 2020-04-05:
@vcapra1 Thanks for the heads-up. Would you be willing to show pics of your aluminum parking post setup? Also, do you remember if you printed the wedge plate with 0.1mm layer height?

On my end, slowing down the locking procedure, pushing into the coupling before engaging it (that's a feature that's now present in the current toolchange macros), and updating the parking post design has basically eliminated this issue on my machine, but that's not to say that there could be something design-related that is making this issue crop up. Any more info on your setup would greatly help, and thanks for chiming in!

vcapra1, 2020-04-05:
I printed the wedge plates with 0.2mm height, so using 0.1mm height might help too.  In my experience, 0.2mm layers are stronger than 0.1mm layers (not sure if that also applies for this piece though, and that could be just a configuration issue that I haven't figured out yet too).

Here are photos of my parking post setup:

[image]

[image]

I have thin blue shims under the blocks to adjust the height, which is a disadvantage of making this out of a single block of aluminum, but I've found the sturdiness has helped a lot.  I haven't tried the new printed parking post design, so that may also work just as well.

HaythamB, 2020-04-05:
@vcapra1 you just gave me an idea. Another approach would be some cast corners and a threaded dowel rod. Found them cheap in 6x60 here:
https://www.boneham.co.uk/Products/dowel-pins/tapped-metric-dowel-pins

The dowel would fasten to a screw through the cast bracket, and the bottom of the bracket goes into a spacer block for correct height. What do you guys think?

Poofjunior, 2020-06-29:
Update on this issue:

I have had a sample run of Delrin Wedge Plates injection molded by some manufacturers in China.
[image]

Prelimininary Repeatability tests look... really good.
[image]

I'll have a bigger batch made and get these in the hands of Jubilee component vendors. They should less than $5 US apiece.

Poofjunior, 2020-07-28:
Ok, here's another image from @erikmcclain 's printer. 

[image]

I think it's safe to close this issue at this point. Huge win for everyone!

closed 2020-07-28 by Poofjunior