machineagency/jubilee #45: Z Homing switch easily damaged

DanalEstes, 2020-02-17:
Any single event that causes Z to not immediately stop will destroy the switch.  These events may occur during initial commissioning, or during regular operation. 

I'm on my 7th replacement Z homing switch

Poofjunior, 2020-02-17:
Yup; I've broken a few myself. Let's start caching ideas here. Here's mine:

I'm really partial to JSteve White's Marionette Probe modification for the Railcore. ([Video](https://www.youtube.com/watch?v=jqDydHZzukI
).) It's retractable and driven by a cable. We could adapt a similar concept inside of the center carriage and tie the probe's retraction cable to the toolchanger Remote Elastic Lock motor. When the tool is unlocked, the probe is exposed. When the tool is locked, the probe is extended.

DanalEstes, 2020-02-17:
I do like the Marionette Probe.  However, if we unlock the tool... well, mine tend to fall off.  Or am I misreading this?

Poofjunior, 2020-02-17:
The Marionette Probe is a cable-driven retractable switch. The idea is that we retract it when tools are locked in the carriage and unretract it when the carriage is bare. That just-so-happens to correspond to  motor movement on the Remote Elastic Lock, so we could attach the cables to the same actuation point.

DanalEstes, 2020-02-17:
Running the Marionette that way would be elegant in an engineering sense.  However, it would rule out probing with a tool mounted.

Poofjunior, 2020-02-17:
Yeah, totally like the appeal, but it may not be necessary or worth doing yet.

You bring up a good point, but it does make me want to ask about conventional use cases. I know z probing with the extruder mounted is how classic 3D printers work, but Jubilee and the E3D Motion System both detach their tools to do the probing. With the capability to remove tools repeatably, is there a reason to still want to probe with the extruder mounted?

### Damage Causes
I have damaged my probe twice, first from scraping it against the bed during toolchanges on the first couple printing layers (fixed by adding a Z hop in the tool change macro) and second from a print fail where the nozzle blobbed up with a goopy mess of PETG that eventually broke off and collided with the Z probe. Can you remember any specific damage causing events?

yvessa, 2020-12-05:
I have just snapped a switch because of a warped layer and eventually, it hit the switch badly.  I'll finish my build and setup it correctly then I might look at it. An idea passing thru my mind, using the twist lock to retract the switch completely while locking the tool and expose it while unlocking the tool.  We could attach a cam or lever arm to the twist lock to do so.  And if moving the switch is not accurate enough, we could drop a guard around the switch using the twist lock while locking a tool. Does it make sense?

javimurcia, 2020-12-08:
Xon 🇦🇺 has reported on discord (https://discord.com/channels/627982902738681876/627988906943578113/785090315346247711) he scraped his Z Switch against the 2 rear retainer screws of the build plate, the ones used as a guide for aligning the removable magnetic plate.

Maybe, we should replace those 2 screws with grub screws so they don't protrude from the print surface.

MichaelPanzer, 2025-02-24:
Hey, Seems like nobody has addressed this in a while. Has anyone considered modifying the carriage to house a bl-touch or something similar?

jackantubis, 2025-02-24:
> Hey, Seems like nobody has addressed this in a while. Has anyone considered modifying the carriage to house a bl-touch or something similar?

Hi, some design add metal pin to avoid killing switch, the metal pin presse on the standard switch and the depth of the pin is set to avoid to kill the switch by too much Z pressure. If you need it I can share the design, check the discord too for this design 👍

MichaelPanzer, 2025-03-17:
It would be very helpful if you could share that deign with me. I already broke one switch and I would like to avoid breaking more.

jackantubis, 2025-03-21:
https://github.com/jackantubis/JUBILEE-Jackantubis-MODS/tree/main/Z%20switch%20mod%20aluminium%20carriage