machineagency/jubilee #53: Toolchanger Lock Actuator Assembly 7 of 15

kelliead, 2020-02-26:
Toolchanger Lock Actuator Assembly sheet 7 of 15

Make it more obvious how the wire harness is assembled. Maybe a diagram with special attention called to the orientation of the switches and how the wires go. 

Also, replace the photo with new one of the completely soldered wire harness. 

[image]

cindygfeng, 2020-02-26:
[image]

Poofjunior, 2020-03-01:
Thanks! I have added this picture, which should leave us in good shape for now until we get proper wiring diagrams everywhere.

closed 2020-03-01 by Poofjunior