machineagency/jubilee #58: Spacer is needed on the inside of the center carriage, around the twist-lock shaft

marioquake, 2020-02-28:
I made a spacer that goes around the twist lock shaft, inside the carriage. it goes between the e-clip and the tool-side, which prevents the twist lock from pulling out from the pulley. It also makes it so i don't have to twist the set screw on the pulley down too far.

if you think about it, the e clip is on the wrong side. its current location prevents the twist lock from going out towards the pulley. But the force is going the other way when you lock onto a tool. the only thing keeping it from slipping out of the pulley is the set screw on the pulley.  this spacer effectively puts the e-clip on both sides inside the center carriage. so it cant slip either direction.

I recommend adding this spacer to your design. It is a simple cylindrical 3d printed spacer, 5.1mm ID, 8mm OD, 27.2mm long.
I ended up filing down the cylinder a tad, and using extra shims to get the fit just right.

Poofjunior, 2020-02-28:
This is a really slick fix to this issue. (Attaching your image from Discord as well.)
[image]
Overall, this looks like a really elegant way to solve this problem.

For reference, the original intention behind the e-clip was to fix the distance between twist lock head and coupling pins to a specific distance. Your solution finishes the task of more cleanly constraining it axially both travel directions. This change also removes the purpose of the Belleville Disc Spring, since it basically completely takes out any axial play.

I would like to merge this, but I also don't want to merge anything on principle without actually validating that it actually works. @marioquake  would you be willing to run your machine in this state and run a test print that requires a few hundred tool changes? The [Dual Color Moai model](https://www.thingiverse.com/thing:1560492) would be a great place to start.

I would also be happy to pilot this fix on my end also.

devcodeio, 2020-03-03:
Question, you had stated that the disc spring serves to add some compression to the twist lock, and that we actually want a bit of axial movement here.  I was thinking it was akin to the spring and thrust bearing that the e3d lock utilizes, adding some compressive force when locking. An assumption that may be incorrect.

Wouldn't the spacer remove that then?   Or am i just misunderstanding the whats at play here?

Poofjunior, 2020-03-03:
@devcodeio good eye. Depending on how it's put together, this setup could remove the axial compression if we also remove the disc spring.

However, you could totally add the disc spring internally where the new spacer would lie to achieve the same effect. 

It's worth testing both with and without the disc spring at this point. (It would be kinda nice to remove it since folks need to buy them in 12-packs.)

Poofjunior, 2020-04-12:
Ok I have removed the Belleville Disc spring and added a spacer on my machine, and this seems to be working.

[image]

Thanks for the input! This is now live on the master branch.

closed 2020-04-12 by Poofjunior