machineagency/jubilee #60: Update BOM to include DIN for components

HaythamB, 2020-03-05:
If we could add a column to the BOM sheet that includes the standardized DIN reference for each of the hardware units, it would make searching for exact replacements much easier for people located in other countries. I'm preparing a sheet with UK hardware suppliers, and including this column wherever I can.

RomanMoisei, 2020-03-05:
it is not very nice to say it now but I had thought about the same thing, not only UK since it should not be a problem to place orders from UK to other countries in the area, or vice versa !!!

HaythamB, 2020-03-06:
Well the BOM's up in the discord for the UK, and I've added in all the nitty gritty DIN/ISO details. It is most likely not a huge priority considering what other issues the repo has logged at the moment. Nice to have sorta thing..

And here's the link in case you'd like it:
https://docs.google.com/spreadsheets/d/15JmLp0KLb-Sfbp9CyB7XrFl7qaECqamUNLdnlR9lXcs/edit?usp=sharing

Poofjunior, 2020-03-07:
Thanks a ton for putting in a bunch of up-front work in this direction!

One option here would be to actually bite the bullet and clearly demarcate two things: a "Bill-of-Materials" and a "Shopping List."

The Bill-of-Materials could be organized by part and contain information like you specify above: DIN/ISO numbers that are related to the parts themselves. 

The Shopping List could be organized by vendor by region for someone to easily thumb through and order everything at once. Ideally, this could be auto-generated by region with some up front work.

If we had a database of parts, organized by DIN/ISO references, we could add vendors by region to these entries. In Solidworks, I can add these DIN/ISO references to the part metadata so that when I export the BOM, they show up as a column. With a bit of scripting, we could feed this Solidworks export to the database and *automatically generate* region-specific vendor BOMs without any human error. Plus--once the database is up, anyone can submit new vendors and regions for a particular part.

Something I currently don't do is export the native Solidworks BOM out from the project. Instead, I make an export for myself and then reorganize the BOM into a shopping list by region, cutting out all the parts that you don't need to buy. I would be more than happy to include the actual BOM in the project exports as well.

Lastly, I can program machines and firmware, but I currently don't have any database skills. If folks are willing to point me in a direction, I'd be happy to start layiing some ground work on this.