machineagency/jubilee #62: Add PT1000 to Extruder BOM

HaythamB, 2020-03-08:
Currently the BMG BOM includes a setup for 2x PT100 temperature sensors, plus associated daughter baords.

The newer PT1000 are now available and don't require a daughter board for Duet, so would be a cheaper option to include.

Poofjunior, 2020-03-17:
Whoa--great mod! I will link to this one for the US
https://www.filastruder.com/collections/all-products/products/e3d-pt1000-sensor

Any notes on how the configuration files change?

HaythamB, 2020-03-17:
Haven't tried it myself, but according to [this Duet forum post](https://forum.duet3d.com/topic/9903/pt1000-temps-are-way-off/4?_=1584482030598), its as simple as M305 Px(heater channel) X500 and you're off. Would need someone who's tried this to chime in..

Poofjunior, 2020-03-17:
Okk, since it's optional, I have added it to the BOM. (I also removed the the thermocouple board, which was an extraneous buy from the getgo if you were using thermistors.)

Thanks for the pro-tip!

closed 2020-03-17 by Poofjunior