machineagency/jubilee #64: broken links in wiki

wabmester, 2020-03-12:
This page: https://github.com/machineagency/jubilee/wiki/Configuring-the-Duet-Controller

This line: "Open the SD card's file system. Overwrite all files in the sys and macros folders with Jubilee's sys and macros folder contents from the repository."

HaythamB, 2020-03-14:
Also, this page: [Guidelines for Extension](https://github.com/machineagency/jubilee/wiki/Guidelines-for-Extension)

Tool interface diagram should redirect to [https://github.com/machineagency/jubilee/blob/master/tools/jubilee_tools/tool_template/tool_interface_diagram.pdf](https://github.com/machineagency/jubilee/blob/master/tools/jubilee_tools/tool_template/tool_interface_diagram.pdf)

Poofjunior, 2020-03-17:
@wabmester thanks! Fixed by @awaer. Also thanks!

@HaythamB thanks! Also fixed.

closed 2020-03-17 by Poofjunior