machineagency/jubilee #67: Mod silicone heater pad to become safer

HaythamB, 2020-03-16:
The standard Keenovo pad wiring is fiberglass sheathing, which tends to quickly wear off from movement and expose the live wires inside. Especially hazardous if we're using an AC pad.

The instructions should reflect replacing the moving components of the heated bed power wires from around the strain relief point to be spliced in with flexible silicone wire, minimum 16AWG /240VAC, 12AWG / 110VAC.

Additionally, consider adding a thermal fuse spliced inline with one of the bed cables, and fastened to the bottom of the heater pad using high-temperature silicone sealant (or equivalent high temp adhesive) to also enhance safety in case of an SSR failure or other AC electrical mishap.  Here's one that I used for my heated bed (you'll need to choose one based on the cutoff temperature for your own build plate). [Thermal fuse for heated bed](https://uk.rs-online.com/web/p/thermal-fuses/7976052/?sra=pstk)

Poofjunior, 2020-03-17:
The current instructions have you sheath the two bed leads together into a nylon sleeve. In that configuration, the leads move as a pair such that they do not abrade each other and they are also protected from abrading anything external.

Additionally, while this may seem unintuitive, the adhesive silicone block on the edge of the bed already serves as a form of strain relief for the two bed leads. What do you envision as an improvement?

Finally, that thermal cutoff is a great safety mod! I'm looking into an easier way to do this in the back of the machine with a bolt-on solution that doesn't require silicone.

HaythamB, 2020-03-17:
I was thinking along the lines of a drag chain, which could possibly be mounted to the bottom of the bed or even screwed on to the Z-axis mount in the rear. On my build the bed is a non-Jubilee subframe where I've attached a drag chain using t-nuts to the rear extrusion.

So its either a couple of blind taps in the bed, or an attachment onto the Z-axis 3D printed part, possibly at the angled face. It could be either 2xM3 or 1xM3, depending on the drag chain used.

Poofjunior, 2020-03-17:
Gotcha. I'd totally take inspiration from pics if you're willing to showcase a couple. For the Z axis, the clearance is pretty tight. One thing to keep in mind is that all fixed or moving components on the bed need to be at the bed plate level or below or they risk snapping the Z axis limit switch.

For thermal cutoff, instead of glue, I'm looking for something to clasp the fuse that you can just bolt-on, kinda like this:
https://www.digikey.com/product-detail/en/keystone-electronics/8140/36-8140-ND/303983

Poofjunior, 2020-03-17:
Actually, this half-cable clamp
https://www.newark.com/ettinger/14-42-341/half-cable-clamp-steel-natural/dp/58Y8078
might work really well with this fuse
https://www.digikey.com/product-detail/en/cantherm/SDF-DF119S/317-1132-ND/1014761

HaythamB, 2020-03-18:
That does look like a great solution, and it comes in a selection of temperatures that people could choose to suit their liking. Only question that remains is if it will remain tight to the bed after repeated movements. Looking at the half clamp tells me it would for 90% of the cases. But I'll stick with sealant to be absolutely sure.