machineagency/jubilee #74: Twist in the CoreXY frame

erikmcclain, 2020-04-07:
Depending on how the machine is assembled it is possible to end up with a twist in the Corexy.  

If there is twist maybe recommendation on how to solve it

[image]
[image]

by Loosening, reclamping the high corners then tightening them back down 

[image]

nexx, 2020-05-29:
@erikmcclain. I'm sorry to intrude, as my enquiry doesn't relate to the Jubilee, but I'm yet to find another person talking about this kind of twist in a CoreXY frame combined with a 3 point kinematic bed. I've been building a RatRig V-Core Pro and adapted it to feature a kinematic bed with 3 independent Z motors. I have exactly this issue where the two free corners have a difference from one side to the other of nearly 1mm despite the 3 Z points being at 0. I know it is not the bed itself, it's 6mm of tooling plate which I know is almost perfectly flat (and I've tried 2 separate sheets), but I've been unable to correct the tilt.

I've checked, measured, double-checked and re-built the frame multiple times now and the results are always the same. I would really appreciate any thoughts you may have on what to try as I'm on the verge of throwing the frame out of the window :P

erikmcclain, 2020-05-29:
Have you joined the jubilee discord?  It would be a good place to have a general discussion.  I would like to see some screen shots of what you have going on.  

That said what I did is put a 12 by 12 polished granit tile on my bed and mesh off of that.  That gave be a very flat surface to start with.  And proved to me that it was a twist in the motion system.   Next I measured the diagonals and I noticed that the front for of the machine had a very slight lean to it.  So I pulled the frame until the I measured identical diagonals.

That took about 1/3 of my twist out.  Next I loosened the low corner indicated by the mesh and the one diagonally across, put a clamp being enough to clamp the top and bottom frames together and tighted everything back up.  That took out about half of the remaining twist. 

Lastly I let the bed heat up for a out 5 min or so with the regular bed surface on.  I homed trammed and meshed.  After that mesh with everything hot for some reason most of the last bit of twist was gone.

nexx, 2020-05-30:
> Have you joined the jubilee discord? It would be a good place to have a general discussion. I would like to see some screen shots of what you have going on.

I don't really use discord I'm afraid.

> That said what I did is put a 12 by 12 polished granit tile on my bed and mesh off of that. That gave be a very flat surface to start with. And proved to me that it was a twist in the motion system. [...]

My latest attempt involved completely rebuilding the frame and using a spare extrusion with 3D printed parts on it as a calibration device. As near as I can tell, every pat of the frame is identically measured and square. That's what makes this so frustrating. I'm also certain that the upright extrusions for the 3 Z axes are both square and true. All the rails were mounted to the extrusions with a set of 3D printed spacers to ensure they're centred.

> Lastly I let the bed heat up for a out 5 min or so with the regular bed surface on. I homed trammed and meshed. After that mesh with everything hot for some reason most of the last bit of twist was gone.

I've not made it that far. I've been doing all measurements with a dial indicator mounted to the X axis. If it's any aid in troubleshooting, I've found that lowering the front right Z point will cause the rear to level out left to right, but then the front is obviously no longer level.

I've also tried two separate tooling plates, both show the same error, so I'm certain they're flat within 0.1mm or less.

Here's a render of the build showing the layout of the axes:
[image]

Prexsys-Beach, 2020-05-30:
Not to be impolite, but perhaps you guys could take this somewhere else :)
I get enough notices about Jubilee stuff :) Good articles on machine
alignment are numerous.  Thank you!

nexx, 2020-05-30:
Apologies @Prexsys-Beach. I won't bother you guys further with unrelated matters. Thank you anyway.