machineagency/jubilee #81: Questions about purpose of shims and shoulder screws in pulley assemblies

crteensy, 2020-04-18:
I have questions regarding the shims and shoulder screws in the pulley assemblies.

**Shims:**
The shims are installed between idler pulleys and printed parts, or between pulleys and plain washers. However, in the "Left Motor Pulley Assembly", there's no shim between the pulley and the "Pulley reinforcement plate for motor". Why?

Also, the standard shims I have looked at do have an outer diameter that is just as large as that of many ball bearing used in idler pulleys. Wouldn't they rub against the bearing's rotating outer ring?

Could 3d printed washers with the right thickness be used instead? Those could also solve the diameter issue - if there is any.

**Shoulder screws:**
I do understand that shoulder screws provide for a more exact centering of the installed parts, and a better surface of ball bearings to press against. They also screw directly into inserts, so that the screw's shaft can press against the insert which results in a correct 90° angle and good transmission of forces without any wiggle. Is that the idea why they're used?

However, shoulder screws are expensive - could normal screws be used and what changes to the design would be necessary?

crteensy, 2020-04-18:
An idea for a replacement: instead of an M4 shoulder screw (25 mm shaft with 5 mm diameter), could we use an M4x30 screw and a piece of 5x0.5 mm pipe (brass would be easy)?

Poofjunior, 2020-04-19:
These are some great observations/questions.

**Shim**
You're right about the shim outer diameter being a bit a bit wider than the outer diameter of the pulley bearing, which would cause rubbing on sliding surfaces. In practice, in this setup, this is ok because the rubbing is happening on a smoother surface (the face of the shim) rather than the opposite pulley faces. That said, a smaller shim would be ideal here to act as a true spacer with no rubbing surfaces, so I spec'ed an 8mm OD shim and added it to the BOM yesterday. (It's also cheaper!) 

As for why there's no shim on the top of the shoulder screw stack in the motor plates, that's because the original motor reinforcement plates were made from Delrin, a very slippy material to begin with. Since the current motor reinforcement plates are printed, it's worth adding them back.

**Shoulder Screws**
In a nutshell, shoulder screws are behaving like shafts in all cases here. True, the outermost diameter of an M5 screw is close to 5mm, but it's not quite close enough (4.826 to 4.976). As a result, the pulley would move around loosely in the shaft. While this probably wouldn't do too much to printer tolerances, it probably would cause a lot of extra vibration and possibly unscrew the screw from its hole without some threadlock. Shoulder screws are dimensioned such that bearings can slide onto the shaft smoothly with little room for any wiggle. One more thing: the bottom surface of the shoulder screw (as you mention) forms a nice contact point with the surface of the connecting plate. Tightening the shoulder screw creates a normal force (a preload) between screw and plate which actually prevents it from unscrewing itself during normal operation.

All-in-all, if you're looking to save money,
* swapping out the electronics for a cheaper controller board
* using unbranded linear rails instead of LDO or Hiwin brand rails
* using generic pulleys instead of Gates brand Idler pulleys
all might be other places to start that wont really sacrifice machine performance.

I hope this helps!

BTW, I'll leave this issue open until I add back the 0.2mm shim on the tops of all shoulder screw stacks.

crteensy, 2020-04-19:
Indeed there's more money to be saved in the parts you mentioned (electronics, rails, pulleys, maybe others too).

I'll have a look around and see what kind of screws I can obtain easily and at reasonable cost. It's not equally easy everywhere.

Poofjunior, 2020-04-29:
Getting back to this a bit late, but you may have some great luck with stainless steel shoulder screws on Aliexpress. I've not tried sourcing shoulder screws from them, but I did notice a good selection.

crteensy, 2020-04-29:
No problem - but the prices are still quite high. Actually the same as for screws sourced locally in DE.