machineagency/jubilee #82: Feet are too short, screw clearances are too tight

BryanHaven, 2020-04-19:
The budget feet are too short to allow heat sinks on a Z-axis stepper motor with a 25mm shaft. Recommend adding at least 10mm.

The mounting screw counterbore clearance is too small for the designated M5 x 10mm long screws.

Poofjunior, 2020-04-21:
[image]
Adding image from Discord for reference.

Just curious, how short is too short? The ones in the BOM are 48mm long. That puts 12mm between the bottom of the motor and the bottom of the foot. I agree that that's not much room for a heatsink, but adding 10mm seems a bit much over, let's say 5mm.

Also: will get the buttonhead screw counterbore hole updated in 2.0.3.

Poofjunior, 2020-06-29:
The Zaxis motors were bumped downwards in 2.1.0, but the feet have now been made taller in 2.1.1.

The total offset between the lowest part of the motor and the floor is now 15mm, which should be plenty of clearance for a heatsink.

closed 2020-06-29 by Poofjunior