machineagency/jubilee #83: Photos for Build Plate Assembly, Z-Axis Assembly Instructions

cindygfeng, 2020-04-23:
Not sure if you want this photo for the Build Plate Assembly because the Heated Pad is already installed, but here it is:
[image]

Also here's a photo for the last sheet of Z-Axis Assembly:
[image]

Poofjunior, 2020-04-23:
OMG those are AMAZING. thats perfect for the docs. Thanks!

cindygfeng, 2020-04-23:
No problem @Poofjunior ! Here is maybe a better version of the second image (I repositioned the top of the spring)
[image]

Poofjunior, 2020-04-23:
They are added! Thanks a bunch; these look great.

closed 2020-04-23 by Poofjunior