machineagency/jubilee #86: Missing shoulder bolt listed in

Prexsys-Beach, 2020-05-01:
The new 25mm shoulder bolts needed in 041_Aluminium_Crossbar_Assembly_Instructions  / Sht 5 of 8 (04/12/2020) are missing from the BOM. The existing BOM lists 90278A374 which is M4 Shoulder Screw, 20mm-l, 5mm-d which is for use with the Delrin bracket.

McMaster Carr 90278A375 is the 25mm version of the same bolt.

Poofjunior, 2020-05-05:
Aye; aluminum corner plates are currently "upgrades" in 2.0.2, so the extra parts that you'll need to buy are listed on Mandala Rose Works' webpage for now, but 25mm shoulder screws should now be the standard. 

I think* this is resolved as of now?
If not, you could be looking at an out of date copy of the bom (perhaps the UK bom?) Here's the main one.
https://docs.google.com/spreadsheets/d/1pRzBQxVzL9c4T9b1RrKvSjlSwJJhJ7NcbSV6iJUv0X0/edit#gid=0

Let me know if this is still an issue.

Poofjunior, 2020-05-29:
This should be a non-issue now that 2.1.0 is released.

closed 2020-05-29 by Poofjunior

Prexsys-Beach, 2020-05-29:
Was not looking at UK version