machineagency/jubilee #102: G10 / Non-standard

arthurwolf, 2020-06-21:
Hello. 

This page : https://jubilee3d.com/index.php?title=TAMV says you are using G10 for a completely out of specs usage. Please please don't! Standards matter, in particular for folks trying to make systems that can run both 3D printer and CNC mill gcode on the same system ( I'm from the smoothieware.org project, which enables this )

We already have a huge issue with the fact that Reprap started using G10 for extruder retract, when it should in fact be a tool to set WCS coordinates ( see the NIST gcode spec ). Now the same Gcode means two completely different things, and we have to jump through hoops all the time thanks to that.

The fact Reprap didn't respect the NIST standards has caused us massive amounts of headache over the years. Please, I beg of you, don't make that issue worse.

You should be able to use G54/G53, and the *normal* version of G10 to do what you want. Please don't create an *additional* out-of-standard G10 use.
If you have a hard time understanding how to use the proper Gcodes here, please tell me and I'll help as much as I can. wolf.arthur@gmail.com

Cheers!

http://smoothieware.org/g10
https://reprap.org/wiki/G-code#G10:_Retract
https://linuxcnc.org/docs/html/gcode.html
https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l1

HaythamB, 2020-06-21:
While that's a very valid input and useful suggestion, it's not something inherent to the Jubilee/TAMV implementation. It's just following the convention from RRF.

Maybe a post on the Duet forums would be in order?

arthurwolf, 2020-06-21:
Thanks, emailed David, hopefully he comes here and gives some more info.

Poofjunior, 2020-06-21:
Oof; it sounds like non-standard use of Gcode commands has caused you a huge development headache. Happy to switch over to conventions that are according to spec if it's possible with the Duet board we're currently working with.

Looking at the Duet [G10 documentation](https://duet3d.dozuki.com/Wiki/Gcode#Section_G10_Tool_Offset_and_Temperature_Setting), it looks like they support a G10 command _with_ and _without_ an L parameter. If the L parameter is omitted, it assumes L=1, which would be a tool table entry.

Can you double-check me here? It looks like, according the LinuxCNC doc, tool offsets can be defined with [G10 L=1](https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l1), but the arguments are limited to tool radius and a few other lathe-only parameters. On a non-spindle machine with multiple tools like a toolchanging printer, a "radius" argument doesn't make sense, but _x_, _y_, and _z_ arguments do make sense.

You mentioned using G53 and G54 for defining tool offsets. Are you suggesting defining one work-place coordinate system per tool? If that's the case, wouldn't we restrict total tool count to 9 tools? If not, can you clarify how we'd go about setting XYZ tool offsets per tool in a way that follows the spec.

Thanks again for chiming in!

arthurwolf, 2020-06-21:
http://smoothieware.org/g10
http://smoothieware.org/g54

More generally :
* http://smoothieware.org/supported-g-codes
* https://linuxcnc.org/docs/html/gcode/coordinates.html
* http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2

> Are you suggesting defining one workspace coordinate system per tool?

No, just set the offset for the current workspace ( in XYZ, there is no
tool radius involved here ).

Poofjunior, 2020-06-21:
Right, but in a multitool setup, we'd likely need unique offsets per tool. Wouldn't that require a custom work offset per tool? i.e: G54 for tool 1, G55 for tool 2, etc. How would we accomplish this with a single workspace coordinate system?

Unless I'm totally misreading something. Are you actually ok with this syntax:
```
G10 L1 P<tool number> X<value> Y<value> Z<value>
```
but not this syntax:
```
G10 P<tool number> X<value> Y<value> Z<value>
```
?
If that's the case, then this something we can just edit in the wiki, as it is in-line with the current duet implementation. If not, can you be a bit more specific about what's incorrect about what's written in the TAMV page regarding G10? If it ends up being a wiki issue, we can fix that. If it's a Duet GCode implementation, then I would gladly vouch for you if you post in the Duet forums. I'd much rather we push for standards rather than reinvent conveniences for ourselves.

arthurwolf, 2020-06-21:
> Right, but in a multitool setup, we'd likely need unique offsets per tool.
You'd store those inside the host program's config, and use it to adjust
the offset of a single WCS when a tool is changed.
Doing otherwise wouldn't match what those Gcodes are for.

> Wouldn't that require a custom work offset per tool? i.e: G54 for tool 1,
> G55 for tool 2, etc. How would we accomplish this with a single workspace
> coordinate system?
Stay in G54, and use G10 L* to modify the offsets for that WCS when
changing tools.

> Unless I'm totally misreading something. Are you actually ok with this
> syntax:
>
> G10 L1 P<tool number> X<value> Y<value> Z<value>
Yes, that's standard.

>  but not this syntax:
>
> G10 P<tool number> X<value> Y<value> Z<value>
And that one isn't standard.

It's all documented:
https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2
https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l20

You stay in G54, and use those to change the offsets whenever you need to (
here when changing a tool ).
This is the correct/standard way of handling this.

Does that clarify the situation?

> ?
> If that's the case, then this something we can just edit in the wiki, as
> it is in-line with the current duet implementation. If not, can you be a
> bit more specific about what's incorrect about [...]

Poofjunior, 2020-06-21:
Ok,
```
G10 L1 P<tool number> X<value> Y<value> Z<value>
```
is supported syntax in the Duet implementation. Happy to be more explicit in the configs and wiki by referring to them as such.

Ok, next question:

> You stay in G54, and use those to change the offsets whenever you need to (
here when changing a tool 

So each time you invoke a tool change, are you saying, you'd follow up with
```G10 L1 P<tool index> X<value> Y<value> Z<value>```? Doesn't that go against the notion of defining these all in advance with a tool table?

dc42, 2020-06-22:
The way RRF behaves in response to a G10 command is:

- G10 with a P parameter but no L parameter sets tool offsets and/or temperatures for the tool specified by the P parameter. This was defined behaviour of RRF before I became involved with it.
- G10 L1 behaves the same as above. This is compatible with LinuxCNC, apart from the option in RRF to set tool temperatures
- G10 L2 sets workplace coordinate offsets in accordance with the NIST GCode standard.
- G10 L20 sets workplace coordinate offsets based on the current tool position. We implemented this version of G10 for compatibility with LinuxCNC.
- G10 with no parameters is a firmware retraction command, compatible with Marlin, Repetier etc.

When you do a tool change, RRF loads the new tool offsets automatically. So no need to follow a tool change with G1 L1 commands.

HTH David

Poofjunior, 2020-07-19:
I have adjusted the docs to suggest more NIST-like usage in the [TAMV docs](https://jubilee3d.com/index.php?title=TAMV#Overview). Thanks all for bringing all this up.

closed 2020-07-19 by Poofjunior