machineagency/jubilee #108: Add setscrew size on page 12 or page 13 of corexy_frame_assembly_instructions.pdf

bcrazycramer, 2020-07-09:
[image]

[image]

bcrazycramer, 2020-07-09:
I also find it confusing to have the toolchanger_carriage_assembly_instructions_draft.pdf file along with this documentation above. Maybe these can be mixed together?

Poofjunior, 2020-08-17:
Ahh, ok. The set screw length (8mm) is called out in the toolchanger_carriage_assembly_instructions_draft.pdf file.

These (and all the instructions) are split up to make the work easier to delegate if you're building the machine with help from more than one person. Separate instructions also make good "stopping" points along the way. But I do think that, out of all the instruction sets, that one is the weakest since you don't finish with one part but rather a bunch of miscellaneous parts.

I'm tempted to leave everything as-is for now since that set screw should already be pre-installed by that point. But before that, do you have any other thoughts now knowing about the setup behind the instructions? Do you have any thoughts on how you might've landed here without the setscrew already preinstalled?

closed 2021-02-03 by Poofjunior