machineagency/jubilee #110: Aluminum Frame not not guaranteed to be electrically connected.

ljones-adaptive, 2020-07-19:
As a user, I would like from an electrical safety point of view to be able to ground the entire frame of the Jubilee 3D printer.

I would be expecting that the tnuts and the corner set screws from the corner brackets, used to assemble the frame, would have suffient bite into the aluminium 2020 extrusion so as to make a complete electrical connection. Thus, allowing the complete frame to be grounded. 

The Problem:
I have found on the hidden corner brackets I sourced from China the set screws have a flat bottom and so do not bite. 
This means only small portions of my frame are actually connected together. 

My solution:
To solve this problem use a continuity tester to inspect the connection between individual 2020 frame extrusions and any found to be non conducting change the set screws for ones with a point until the required connectivity is obtained.

Poofjunior, 2020-07-23:
This is a great point. Just curious: are you using anodized extrusions or unanodized extrusions?

Either way, I think it makes sense to put the builder in a situation where something as simple as a color change doesn't make some machines safer than others. We should test conical or knurled set screws to see if they have enough "bite," and then add them to the BOM.

Thanks for bringing this up!

ljones-adaptive, 2020-07-23:
If I test connectivity using a continuity tester, then I get continuity between the raw ends of the extrusion where a saw has made a cut.
I am unsure what the finish is and can't find where I sourced them from.
If I test conductivity between one of the ends where the cuts are made and any point down the length of the extrusion then I do not get connectivity.
I have solved the problem by using conical pointed set screws for the hidden corner joints.. These bite into the extrusion and make contact. As I change out the set screws, from flat to pointed, I can follow the continuity around the frame until all the extrusions are connected.

So, adding a step to "check continuity" into the bed assembly intructions will probably surfice to increase saftey.

andmiller, 2020-08-12:
The inside brackets here: https://www.amazon.com/s?k=b07vq84gwv&ref=nb_sb_noss looks like it comes with conical set screws, I'll update when I receive them.

andmiller, 2020-08-18:
So the brackets I linked above do indeed come with pointed fasteners and do reliably (from my limited testing) conduct even when only moderately tightened. I have tested with 4 lengths of extrusion that are not touching, but only connected by the brackets and fasteners and get good continuity from the first to the fourth.

Poofjunior, 2020-08-18:
Thanks for checking and following up!
So far, I've listed cup point set screws in the BOM as optional replacements for the ones that come with the L brackets. My hope is that these should cut through the anodization without needing to check, but this is untested.

Big picture here. Long term, my intent is that "blindly" ordering the listed parts in the BOM produces a functional safe machine without the assembler needing to know too much about how certain things work (unless they want to go deeper). With that said, I'm hoping I can add a note about continuity checks *only if you ordered different L brackets.*

Poofjunior, 2021-03-16:
Ok, cup point set screws are now (1) noted in the BOM, (2) mentioned in the [L-Bracket Assembly](https://github.com/machineagency/jubilee/blob/master/frame/assembly_instructions/frame/inside_corner_bracket_prep.pdf), and (3) included in Filastruder's fastener kit by default. (Filastruder also removes the flat-point set screws from the L-Bracket bags.)

Cone point are also listed as an option in the instructions too.

I think it's safe to close this one now. Thanks for the sanity checks during your assembly process @andmiller and the heads-up from the default included screws @ljones-adaptive !

closed 2021-03-16 by Poofjunior