machineagency/jubilee #112: Misconfigured System Breaks X Axis Limit Switch

Poofjunior, 2020-07-29:
If the machine is misconfigured or endstops are not plugged in, the machine can damage the X limit switch while homing as it continues to travel into the switch, crushing it.

It would make sense to create a situation where, even if the switch doesn't trigger, the machine hard-stops, rather than causing damage to itself that requires the user to service it. (The Y Axis is like this. Rather than break the switch, it hard-stops into the frame.)

coldelectrons, 2020-10-23:
I suggest a modification of CBP-05: extend the plastic past the switch by 3.0 to 3.2 mm. Based on my digital calipers and one of those switches, that is how far the plunger still extends when triggered. The switch itself can remain in the exact same position.
I myself have now clobbered two of those switches getting my Jubilee moving for the first time.

Poofjunior, 2020-10-24:
Thanks for the rec @coldelectrons ! This is one possible solution. We'll have to check to see if we need to modify the corresponding tab where the switch triggers, so there's a low-stress surface to act as a hard-stop.

Poofjunior, 2021-06-03:
Ok, here is an in-progress solution in related issue: https://github.com/machineagency/jubilee/issues/180

Poofjunior, 2021-06-03:
Ok, instructions are updated.
This part is now in the print list:
https://github.com/machineagency/jubilee/blob/main/frame/fabrication_exports/3d_printed_parts/frame/x_axis_hard_stop.STL

And I think we're a lot better now than where we were!

closed 2021-06-03 by Poofjunior