machineagency/jubilee #130: Duet3 resume.g contain tfree0 gcode

olby, 2020-11-16:
Tried to resume my print just to see my tool drop.

Can we replace the duet 3 beta resume.g with the code from the rrf3 for duet 2?

olby, 2020-11-16:
Oh, the pause.g of the duet3 contains the resume macro. Let me fork and create a pull request once i get it running.

Poofjunior, 2020-11-16:
Yes. Yeah, both the Duet3 configs and the folder they're living in need some rework to be a bit clearer.

If you're up for making a pull request, let's work off the dev/jubilee_2.2.0 branch where I've partially refactored the [folder structure](https://github.com/machineagency/jubilee/tree/dev/jubilee_2.2.0/software/duet_config_files) for the configs. Thanks for giving this a go!

olby, 2020-12-05:
Pull request #138

closed 2021-04-21 by Poofjunior