machineagency/jubilee #139: Possible security breach on Jubilee wiki site

mhipszki, 2020-12-14:
There seem to be unwanted changes on the Jubilee wiki looking at the change log:

[image]

The _Main page_ content also looks weird...

This is a fantastic project and thought it'd be useful to flag that @Poofjunior  🙂

HaythamB, 2020-12-14:
Yup looks like its been hacked. I'll see if I can disable new users for the time being until Joshua is on US daytime!

HaythamB, 2020-12-14:
@Poofjunior I think you'll need to install https://www.mediawiki.org/wiki/Extension:UserMerge

I'm deleting all the spam pages, already restored the wiki contents page-by-page to remove all the bots' updates.

HaythamB, 2020-12-14:
And looks like we started getting spam users as far back as December 3rd.

HaythamB, 2020-12-14:
2549 pages deleted so far..

mhipszki, 2020-12-14:
😱  omg

Prexsys-Beach, 2020-12-14:
Why can’t we have nice things? :)

Poofjunior, 2020-12-14:
Oof; thanks for bringing this up @mhipszki , and thanks for axing the spam pages @HaythamB . 
I only noticed the front page last night and promptly changed it back and blocked the user.

Yeah; it looks like we've hit a critical mass where we might have to add some flavor of captcha to keep the spam at bay longer term.

HaythamB, 2020-12-14:
Ready to be marked as done, and I've created a relevant issue for addressing this in the longer term #141

Poofjunior, 2020-12-14:
Thanks a bunch @HaythamB !

closed 2020-12-14 by Poofjunior