machineagency/jubilee #140: Adapting Jubilee to 3030 extrusions / MGN-xx-H carriages

mhipszki, 2020-12-14:
So I've got access to 3030 extrusions and `MGN-xx-H` rails and I'd like to adapt the designed parts to them.

I noticed that the Jubilee CAD drawings are beautifully organised under the `frame/cads/Solidworks` folder and wonder if it'd be possible to modify or contribute to those parts with a CAD program other than Solidworks?

Unfortunately I don't have access to Solidworks neither I could run it on my Mac. However I noticed I could use the exported STEP file and modify the 3D printable parts for my Jubilee, but then was also thinking there could be other people already having access to these components e.g. they'd like to modify their HEVO, reusing its parts... so might as well publish it for them?

I'm aware though that supporting a 3030 frame would result in duplicate work when it comes to design improvements involving changes in 3D printed / machined parts etc. but maybe it's something worth discussing 🙂

Poofjunior, 2020-12-14:
Yeah; one way to make progress on a 3030 frame would be to start from a Fork of the project and then develop your changes such that they stay in line with the folder structure of the project. You could make a new folder in frame/cads. And if your Fork is public, then folks can use your files as-is, even if you don't have instructions for them or a complete BOM. From here; merging would be a matter of simply merging in your new folders/files.

For now, i think STEP is an acceptable file format if you want to save CAD files that way. Big picture, a competitive, accessible CAD platform is something we could all benefit from, but we just have to do our best for now.

I admit that the bar is high for frame changes. Not because I'm emotionally tied to the design, but because it's tied to a BOM and assembly instructions that make it easier to consistently reproduce from various folks. Having one design also makes it easier for folks to debug each other's setups via Discord. Getting a substantial frame change merged in would probably need to come with instructions and a BOM of some sort so that we could help folks build it to the same level of granularity as they can with the original.

One philosophical question that's been on my mind is: *if the frame is different, is it still Jubilee?* I don't really have a hard answer yet, but I'm leaning towards no. This is because a different frame will have different properties that make it harder to tap into community knowledge to help with bringup/debugging issues that arise.

All that said, I don't want to discourage you from exloring this--you totally should! One thing we can do is, as you make progress and get it online, we can link to it in the wiki. Right now we have a [Community Mods](https://jubilee3d.com/index.php?title=Community_Mods) page, which is like a catch-all for tweaks that people have made. It would be great to develop this more so that other more mature designs get their own page/pages on the wiki to help guide folks who want to explore making changes. It'd give them a sense of what works and what has been done before.

Let me know what you think, and feel free to chime in on Discord too.

mhipszki, 2020-12-18:
Thanks @Poofjunior for being open and kindly sharing your thoughts!

I'm happy to fork the project and work out the changes on the STEP file, however you are absolutely right in that changing the frame would have a knock on effect on most parts and eventually on the BOM and the instructions. Initially when I was thinking how it could be done I didn't take that into consideration.

I totally agree with all that's said and that the intention to keep a single, reproducible and "evolving" version of Jubilee is a very important aspect of a project like this. Still, I'm keen to explore the "3030 frame" route and am willing to spend time on working out the details. I guess the other option would be to just toss the parts I've already got in the drawer and buy all that prescribed in the Jubilee BOM but that would be less fun! 😄

On the philosophical level regarding the naming and relationship, I also understand your thinking and even if I worked out a solution for the public using a 3030 frame I wouldn't be sure myself if it's still a "Jubilee" or not but I'm happy to rely on your recommendation. Let's cross that bridge when we get there!

The plan is to spend time on this project during the coming holidays, so let's see how far I can get. I've joined Discord already and will get in touch there.

Regarding the `competitive, accessible CAD platform`, what do you have in mind?

Poofjunior, 2021-03-16:
Whoop--I'm coming at this pretty late. I'll pop in my last reply, and then we can continue the chat on Discord if you like.

As far as an accessible cad package goes, I think FreeCAD and Alibre are both worth exploring. FreeCAD is open source; Alibre is a one-time purchase (with a significantly cheaper annual upgrade purchase.) Both have some sort of "Assembly" and "dimensioned drawing" application, which something like Jubilee uses heavily to manage the model's complexity and make instructions. As far as what to pursue, I'd be open to what folks think about either!

closed 2021-03-16 by Poofjunior