machineagency/jubilee #145: Duet 2/RRF3 Config.g mixes up PT100/PT1000

FL-CDF, 2021-01-09:
In the config regarding the extruder temperature sensor there is a mix up betwin PT100&PT1000

The current code also didn work with the stock E3D PT100.
The currend code in config.g : M308 S1 P"e0_temp" Y"thermistor" T100000 B4725 C7.06e-8     ; PT1000-style E0 temperature sensor V6

Corrected code for the BMG tool with V6: 
M308 S1 P"e0_temp" Y"thermistor" T100000 B4725 C7.06e-8     ; PT1000-style E0 temperature sensor V6

Poofjunior, 2021-01-09:
Much appreciated! I'll squeeze this into the next release.

Poofjunior, 2021-03-16:
Ok, so apparently the R parameter is autospecified in Duet3, so we can just leave it out.
https://duet3d.dozuki.com/Wiki/Connecting_thermistors_or_PT1000_temperature_sensors#Section_The_M305_M308_R_parameter
(Oof; that threw me on a side quest.)

Poofjunior, 2021-03-16:
Ok; this is now up-to-date. Thanks for the heads-up!

closed 2021-03-16 by Poofjunior