machineagency/jubilee #167: Add a temperature Preheat to the Bed before probe in Beginning GCode

edskull, 2021-02-17:
Add M190 S[first_layer_bed_temperature] (or equivalent arguement if slicer allows) before the probe to reduce errors due to bed expansion

edskull, 2021-02-17:
Example:
G91 ; relative moves
G1 Z1 F900 ; raise tool 1mm
M140 S[first_layer_bed_temperature] ; Set Bed Temp
M104  S[first_layer_temperature_0] T[current_extruder] ; Set first tool temp
G90 ; absolute moves
T-1 ; Make sure nothing is parked on the carriage
G0 X150 Y150 F10000; Move to the center of the print area
M558 F500  ; Set the probing speed
M190 S[first_layer_bed_temperature] ; Wait for Bed Temp
G30 ; Do a single probe
M558 F50 ; Set a slower probing speed
G30 ; Do a second probe
G29 S1 ; Load Heightmap.csv
T[current_extruder]

Poofjunior, 2021-02-17:
Yeah; if this is supported in the slicer, we should definitely add a check like this. The bed probing is kind of a hack to fix this issue as-is, but waiting till the bed comes up to temperature before reprobing is a better approach.

edskull, 2021-02-21:
Thinking further on this, and im not sure if this is possible in firmware due to limitations, but one could use the Material information for the toolplate to calculate the difference between printing temperature and probing temperature. https://www.upmet.com/sites/default/files/datasheets/mic-6.pdf going from 20C to 80C for example would give a difference of around 0.036mm, negligible for most but i suppose for people doing high temp stuff this might become significant