machineagency/jubilee #189: Baby Bullet, wiring diagram

kelliead, 2021-11-17:
Need a diagram for the wiring on the tool side of the baby bullet, the wire colors coming out of the orbiter (yellow, blue, green, red) don't match the wire colors in the default wiring harness diagram (black, green, blue, red). 

Baby Bullet tool may need its own diagram for the tool side wiring, or this info added to the default wiring diagram! 

Alternatively, a description on the wiki for how to determine what order the wires coming out of your motor should be, might be helpful.

kelliead, 2021-11-17:
if a wiring diagram is created for the baby bullet, also noting that the dual part fans are wired in parallel

kelliead, 2021-11-18:
Also, the default wiring diagram for Duet 2 machines is hidden in the Bondtech extruder instructions. Is it comparable for the Baby Bullet tool?

Poofjunior, 2021-12-02:
Ok, here's what I can do:
[image]
I'll put a snippet of the LDO orbiter tool wiring in that diagram so that folks can see the full-picture setup from the boards to the extruder.

For Fan wiring, I still don't have any connectors I like, but I'm still looking.